arduino/compile-sketches #2: Import relevant commit history from experimental repository

per1234, 2020-09-09:
The initial development work on the action was done in a repository used to host actions while they are in their unstable, experimental stage of development. The valuable commit history should be imported into the dedicated repository for the action.

---
All commits relevant to this action were filtered from the full history of that repository using the excellent [`newren/git-filter-repo`](https://github.com/newren/git-filter-repo):
```
git clone https://github.com/newren/git-filter-repo
cd git-filter-repo
# checkout latest tag
git checkout $(git describe --tags `git rev-list --tags --max-count=1`)
export PATH="$PATH:$PWD"
cd ..
git clone https://github.com/arduino/actions
cd actions
git filter-repo --path LICENSE \
                --path .github/workflows/libraries_compile-examples.yml \
                --path .github/workflows/spell-check.yml \
                --path etc/codespell-ignore-words-list.txt \
                --path libraries/compile-examples --path-rename libraries/compile-examples/: \
                --message-callback 'return message.split(b"libraries/compile-examples:", maxsplit=1)[1].strip()[:1].upper() + message.split(b"libraries/compile-examples:", maxsplit=1)[1].strip()[1:] if message.startswith(b"libraries/compile-examples:") else message.strip()'
```

---
Since it's not possible to add this commit history to the repository via a pull request, I have pushed it to a development branch.
https://github.com/arduino/compile-sketches/tree/per1234/development-initial-import

This development branch should be reviewed. Once it is approved, I'll push it to the primary `main` branch.

I have also submitted a PR to update the CI workflows. You can see from the CI runs on that PR that all checks are still passing after the import:
https://github.com/arduino/compile-sketches/pull/1/checks

per1234, 2020-09-09:
Hi @aentinger. When you get a chance, would you mind taking a look at this? TIA!

aentinger, 2020-09-09:
As far as I can tell this all looks good :+1: Is this issue mainly documentation or am I misunderstanding something?

per1234, 2020-09-09:
Thanks!

This issue in only intended to be a way to follow Arduino's standard "propose, review, approve, implement" procedure in a situation where it's not possible to use the tools GitHub provides to facilicate that process in PRs.

closed 2020-09-09 by per1234

aentinger, 2020-09-10:
Roger that ;) Consider it review and approve then ;)

aentinger, 2020-09-10:
And good work of course :rocket: