arduino/compile-sketches #22: Support for ESP32

ibaranov-cp, 2021-04-14:
Hello,

Considering that the ESP32 folks seem to have built their own here:
https://github.com/espressif/arduino-esp32/actions/workflows/push.yml

Is the official arduino compile-sketches going to support ESP32 as well? Considering there is some special compilers/tools needed.

I'm asking as I'm trying to get this going here:
https://github.com/Blueprint-Foundry/Shelly/actions/workflows/compile-test.yml
However I am fairly new to Github actions, and don't want to dig in the wrong direction if ESP32 is fundamentally not supported yet. 

Cheers :)

per1234, 2021-04-14:
Hi @ibaranov-cp. Thanks for your interest in this project!

Yes, all Arduino boards platforms are supported. This is a general purpose tool designed specifically with this sort of flexibility in mind.

You can see an example of ESP32 being used with the action in this workflow for the Arduino_ConnectionHandler library here:
https://github.com/arduino-libraries/Arduino_ConnectionHandler/blob/master/.github/workflows/compile-examples.yml

There is something fairly unique about the ESP32 platform which requires some special treatment for installation, both locally as well as in the GitHub Actions runner. We are accustomed to Arduino Boards platforms installing all necessary dependencies automatically during the Boards Manager installation. However, the ESP32 platform does not. The platform has a dependency on the `pyserial` Python package, and the user is expected to install this themselves. This did require an adjustment of the configuration of this action's virtual environment, but that has been done already some time ago (https://github.com/arduino/compile-sketches/pull/15) and so it is now no problem to install the external dependency, as you can see done here:
https://github.com/arduino-libraries/Arduino_ConnectionHandler/blob/0.6.3/.github/workflows/compile-examples.yml#L110

If you run into any problems using the action with ESP32, I can help out over on the relevant pull request (https://github.com/Blueprint-Foundry/Shelly/pull/3).

closed 2021-04-14 by per1234

ibaranov-cp, 2021-04-14:
Awesome, thank you for the super quick and detailed response, and the direct PR!
Merged, and very much appreciated @per1234