arduino/compile-sketches #25: How to get the compiled HEX?

TomasHubelbauer, 2021-04-29:
Hello, is it possible to get the compiled HEX from the sketch?

per1234, 2021-04-29:
Hi @TomasHubelbauer. Yes, you can do this.

The action's [`cli-compile-flags`](https://github.com/arduino/compile-sketches#cli-compile-flags) input allows you to pass arbitrary flags that will be used in the `arduino-cli compile` command that compiles the sketches. You can learn about all the available flags here:
https://arduino.github.io/arduino-cli/latest/commands/arduino-cli_compile/#options

There is an example of a configuration that uses `arduino-cli compile`'s `--export-binaries  ` flag` here: https://github.com/arduino/compile-sketches/pull/18#issue-569003805

closed 2021-04-29 by per1234

TomasHubelbauer, 2021-04-29:
I tried this:

```yml
    - name: Compile the sketch
      uses: arduino/compile-sketches@v1 # https://github.com/arduino/compile-sketches
      with:
        sketch-paths: |
          - .
        cli-compile-flags: |
          - --output-dir="$PWD"
```

But it did not work. I am not sure if I need to use `--build-path` maybe or instead of `$PWD` use `.`, in any case, I found this worked:

```bash
cp /tmp/arduino-sketch-*/sketch-name.ino.hex sketch-name.hex
```

I added it to a script step and it works well enough for now until I figure out `cli-compile-flags`.