arduino/compile-sketches #59: output compile time for each sketch

2bndy5, 2022-04-17:
I had a thought that might be useful for workflow maintainers: Output the time it takes to compile an individual sketch. It might also instigate optimization for sketches that take longer to compile, but that is subject to sketch author's intentions.

There's currently no way of looking at a completed workflow's log and determine which sketch(s) cost more minutes (rather important for private repos). Compared to PlatformIO's CI capability, each sketch is compiled separately as individual jobs, thus the compile time is displayed as part of the job's runtime (not an exact measurement but helpful). Furthermore, PIO output does include how long it took to compile a sketch. Since this action compiles several sketches in 1 job, the job's runtime is cumulative and less representative of an individual sketch's compile time.

2bndy5, 2022-04-17:
It could be argued that this idea belongs in the arduino CLI offerings. I don't see a way to do this on that end though.

per1234, 2022-04-17:
Thanks for your suggestion @2bndy5.
 
>It could be argued that this idea belongs in the arduino CLI

It could certainly be done in Arduino CLI, but I think the feature has less value for those who use Arduino CLI directly because It is fairly easy for them to time the `arduino-cli compile` command duration on their own (e.g., using [the `time` command](https://man7.org/linux/man-pages/man1/time.1.html)).

As you point out, the workflow step that runs the action will often be used to compile multiple sketches. Even if you configured your workflow to only compile one sketch per job, the step run time displayed in the logs includes the dependencies installation overhead. So this is something that is not easy to achieve from outside the action (the raw workflow run logs provided by GitHub Actions do contain timestamps, so it is possible to get the compilation times, but certainly not convenient).

PR #61 (2022-04-18): output compilation time for each sketch
resolves #59 

This is a very simple approach. If I knew the code-base better, then I'd decorate a function (like `run_command()`), but the basic aim for #59 is only meant for timing compilation of sketches instead of other action operations.
closed 2022-04-18 by per1234