arduino/compile-sketches #60: could a runner's cache be supported?

2bndy5, 2022-04-17:
This question is what made me think of the #59 suggestion because it would aid in reducing the action's runtime. My intention is to make use of `actions/cache@v3`.

Initially, I thought to install the python src of this action as a step then execute the python code as another step, but it looks like that would be skipping the action's setup script.

So, this question seems dependent on where the cache-able assets exist (in the docker container's FS or the runner's FS). When I say "cache-able assets", I mean the Arduino cores (builtin or externally fetched like the ATTinyCore). I don't think that caching the size-delta reports would be applicable/helpful. Maybe the ArduinoCLI tool could also apply to the cache...

per1234, 2022-04-18:
Hi @2bndy5. I made some efforts to set up caching in all our standardized GitHub Actions workflows in preparation for widescale deployment to Arduino's repositories last year. However, I don't know a lot about the subject matter, was told that it was not worth the effort, and did not find information in my researches that would allow me to justify the significant increase in complexity and maintenance burden it added to the workflows. So I reverted it. I documented everything I know on the subject in that commit here: https://github.com/per1234/.github/commit/f1cadde71474887dc6a5a5c38411364496d27c04

I am certainly interested in the idea of making meaningful improvements to the efficiency of the action or workflows. I would give careful consideration to a well researched and defended proposal for any of the following:

## Caching within the action

Ideally this would be done in a manner that separates GitHub Actions-specific operations from what is mostly not a GitHub Actions-specific Python script.

## Documentation of caching in workflows using this action

This would be added to the action's documentation to serve as a reference to its users.

## Caching in Arduino's standardized workflows

We have a collection of standardized workflows which are maintained in dedicated repositories:

- https://github.com/arduino/tooling-project-assets/tree/main/workflow-templates
- https://github.com/per1234/.github/tree/main/workflow-templates

---

>in the docker container's FS

This action does not use a Docker container. It did during the earlier phase of development when that was the only option, but we decided to switch it to a ["Composite Action"](https://docs.github.com/en/actions/creating-actions/creating-a-composite-action) (originally called "composite run steps action") after GitHub made this option available. The action runs directly in the GitHub Actions runner machine.

2bndy5, 2022-04-18:
@per1234 Its always a pleasure to talk "shop" with you! I wan't aware of the previous efforts. BTW, some of that commit's links (in the description) are now 404. 

> since GitHub is on AWS and arduino.cc is on AWS, which makes the download nearly instantaneous

This surprised me. I hadn't considered the source of the assets when opening this issue.

> The action runs directly in the GitHub Actions runner machine.

After looking closely at the action_setup,sh script, I think there's nothing stopping me from executing this action's python src code as an individual step (given that most input options - including venv activation path - is a env var or an identified step output.

> [image]

There's no need to burden this exploration on you alone. I can start some rudimentary experiments with RF24 lib and report my findings. Full disclosure, I'll probably just execute the action as a python script with env vars and pseudo step output vars to make use of `actions/cache@v3`.

It would really help measure the benefit of caching if/when #59 is resolved. I'm unsure of the contributing standards here, but I'm willing to contribute a simple solution for that.

per1234, 2022-04-18:
>some of that commit's links (in the description) are now 404.

Yeah, it is not a good commit message. I actually just did a copy/paste of a summary of the situation I provided someone I was asking for an opinion on whether the caching efforts should be reverted.

>I'm willing to contribute a simple solution for that.

Contributions are welcome!

per1234, 2026-03-10:
From https://github.com/arduino/compile-sketches/issues/473:

> Currently, cold starts can take around 50 seconds, and full builds with multiple sketches can take 4–5 minutes. By adding caching for dependencies and moving from **Poetry** to **[uv](https://docs.astral.sh/uv/)**, I was able to reduce cold start times to ~25 seconds and some full builds from ~4.5 minutes to ~1.5 minutes.
>
>[...]
>
>**Example runs:**
>- Original action (4 minutes and 30 seconds): [Run link](https://github.com/arduino/app-bricks-examples/actions/runs/22726425849/job/65902702551) 
>- Improved action with caching & uv (1 minute and 40 seconds): [Run link](https://github.com/abstract-333/app-bricks-examples/actions/runs/22740836330/job/65954663742) 
>
>**Testing the improved version:**  
>Maintainers can try the early-stage version here:  
>`abstract-333/compile-sketches@v1.1.0`

Hi @abstract-333. In order for the action to provide accurate results, it is essential that the action installs the exact dependencies specified by the user. The approach you have implemented:

<https://github.com/abstract-333/compile-sketches/blob/a89d6bbefef4ce3c00dcdbbb82ccc90e2c021693/compilesketches/compilesketches.py#L398-L406>

```python
        # Simple check: if the FQBN directory exists in the packages folder
        if self.enable_cache:
            fqbn_parts = self.fqbn.split(":")
            if len(fqbn_parts) >= 2:
                vendor, arch = fqbn_parts[0], fqbn_parts[1]
                check_path = self.board_manager_platforms_path.joinpath(vendor, "hardware", arch)
                if check_path.exists():
                    print(f"🚀 Cache Hit: Platform {vendor}:{arch} found. Skipping installation.")
                    return
```

only checks that a platform with the specified identifier is installed when determining whether to use the cache. It doesn't verify that the cached platform is the version specified by the user.

This action supports installing arbitrary development versions of dependencies in addition to the release versions. So there are a lot of corner cases to consider in determining whether it is appropriate to use the cache. I would say that the cache should only be used if the set of dependencies specified by the user are identical to those specified when the cache was established.

Another thing to consider is that we support installing the latest version of a dependency. In this case, the reuse of cache that is [as much as seven days old](https://docs.github.com/en/actions/reference/workflows-and-actions/dependency-caching#default-limits) would not be ideal, since the user likely wants their CI to quickly make them aware of breakage that was introduced by a new release of a dependency.

> I was able to reduce cold start times to ~25 seconds and some full builds from ~4.5 minutes to ~1.5 minutes.

That number comes from two distinct changes. Please profile the impact of the cache alone, without your other proposed change of migrating from Poetry to uv. Provide that number in a comment here.