arduino/compile-sketches #85: compile-sketches uses gnu++11...

FredM67, 2022-11-21:
### Describe the problem

My sketches need to set gnu++17 instead of gnu++11 in the platform.txt file.

How to do the same when compiling on GitHub?

### To reproduce

Compile some sketches with a c++17 feature, not available in c++11.

### Expected behavior

It should be possible to overwrite the c++ version used to compile.

### 'arduino/compile-sketches' version

latest

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/compile-sketches/issues?q=)
- [X] I verified the problem still occurs when using the latest version
- [X] My report contains all necessary details

per1234, 2022-11-21:
Hi @FredM67. You can add arbitrary flags to the `arduino-cli compile` command used to compile the sketch via the action's `cli-compile-flags` input:

https://github.com/arduino/compile-sketches#cli-compile-flags

You can learn the available flags from the Arduino CLI documentation:

https://arduino.github.io/arduino-cli/latest/commands/arduino-cli_compile/#options

In this case, the flag of interest is `--build-property`. Depending on which boards platform you are using, you may be able to use the `compiler.cpp.extra_flags` build property to add additional flags to the compilation command. So if you wanted to add the `-std=gnu++17` flag to the compilation command, you would add this to your workflow under the `with` key of the step that runs the `arduino/compile-sketches` action:

```yaml
cli-compile-flags: |
  - --build-property
  - compiler.cpp.extra_flags=-std=gnu++17
```

Please note that usability of build properties is dependent on how each boards platform author has configured their platform. Providing customization of the compilation command via `compiler.cpp.extra_flags`, etc. build properties is a common convention established by Arduino's official boards platforms, but there is no guarantee of that. Some platform authors have also unnecessarily used the properties for their own purposes (https://github.com/arduino/arduino-cli/issues/846). So this is an advanced usage and you should make sure to study the boards platform to make sure it supports this.

closed 2022-11-21 by per1234

FredM67, 2022-11-21:
> ```yaml
> cli-compile-flags: |
>   - --build-property
>   - compiler.cpp.extra_flags=-std=gnu++17
> ```

Thx a LOT for the example.... I've tried every possible combinaison, but couldn't find the right way to add this flag.
I think, a sample should be added to the doc !

per1234, 2022-11-21:
You are welcome.

I completely agree that the general usage of this input is not adequately documented. A pull request adding an example. I would not use something so complex as `--build-property`, but I do think demonstrating the correct way to pass flag that have an argument is important. So I think the best candidate for the example would be `--output-dir`