arduino/compile-sketches #129: Ability to exclude specific examples in a matrix build

tfry-git, 2023-04-02:
### Describe the request

Almost all of the (many) examples in my library are meant to work on all supported boards. However, there are a few exceptions that are specific to a certain platform.

I am aware that I can define a "positive" matrix to the effect of "sketches A and B should be compiled for all platforms, but sketch C should only be compiled for AVR". However, this approach is a) pretty verbose, where dozens of examples are involved, and b) just waiting to get outdated as new examples are added/removed. 

What I would like to be able to do, therefore, is to exclude certain sketches from a path, i.e. "all sketches should be compiled for all platforms, except sketch C should be excluded on RP2040". I'm thinking of perhaps a "exclude-sketch-paths" to be applied after "sketch-paths".

### Describe the current behavior

```sketch-paths``` allows to define a (recursive) positive whitelist of examples to compile, but there does not seem to be an ```exclude-sketch-paths```` counterpart.

### 'arduino/compile-sketches' version

v1.1.0

### Issue checklist

- [X] I searched for previous requests in [the issue tracker](https://github.com/arduino/compile-sketches/issues?q=)
- [X] I verified the feature was still missing when using the latest version
- [X] My request contains all necessary details

per1234, 2023-04-02:
Hi @tfry-git. This capability is already provided by the GitHub Actions workflow framework:

https://docs.github.com/en/actions/using-workflows/workflow-syntax-for-github-actions#jobsjob_idstrategymatrixexclude

closed 2023-04-02 by per1234

tfry-git, 2023-04-03:
I admit that I wasn't aware of matrix exclude, but I don't think that helps (or I am too dense, which would turn this into a request for an added example in the documentation):

The problem is that sketch directories are matched recursively (for a good reason!), but I want to exclude only a sub-path (i.e. not something that can be had on the level of the matrix). I.e., I want to write:

```
   sketch-paths: |
      - examples
   exclude-sketch-paths: |
      - examples/Just_this_one
```

(Which, coming to think about it, is even independent from matrix builds, in the first place).

TheNitek, 2024-06-23:
@per1234  any chance to un-decline this? I am in the same situation as the poster and it would come very handy to exclude a specific example for some of the platforms!