arduino/compile-sketches #142: Ability to prep build environment without building

witnessmenow, 2023-05-14:
### Describe the request

For my use case I need to be able to modify files in an installed library before the build takes place.

For example, [tft_eSPI](https://github.com/Bodmer/TFT_eSPI) is a popular library for different displays, but it requires the user to update a `User_Setup.h` file inside the library. From the library Readme:

> The screen controller, interface pins and library configuration settings must be defined inside the library. They can NOT be defined in the Arduino sketch. 

In order to use the library I would need to:
- install the library
- copy a custom `User_Setup.h` into the library
- compile the code

Unless I'm missing something (And I could be, i'm pretty new to Arduino CLI and Github actions), I don't think that's possible with the current setup?

### Describe the current behavior

I don't see how to achieve this with current options/flags

### 'arduino/compile-sketches' version

latest

### Additional context

Thanks so much for the work on this, I've already started to use for some of my libraries to test examples!

### Issue checklist

- [X] I searched for previous requests in [the issue tracker](https://github.com/arduino/compile-sketches/issues?q=)
- [X] I verified the feature was still missing when using the latest version
- [X] My request contains all necessary details

witnessmenow, 2023-05-14:
I am able to achieve it by building twice, but its a little wasteful!

```
# Fake Arduino Code
      - uses: arduino/compile-sketches@v1
        name: Fake Build to install libraries
        with: 
          fqbn: "esp32:esp32:esp32"
          platforms: |
            - name: esp32:esp32
              source-url: https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json
          # No need to specify ESP-libraries as these are installed with the platform (on the line above). 
          # Downloading SpeedyStepper from source to get correct casing on import of Arduino.h (spelled arduino.h in the version from Arduino Libary Manager)
          libraries: | 
            - name: ImageFetcher
              source-url: https://github.com/witnessmenow/file-fetcher-arduino.git
            - name: WiFiManager
            - name: ESP_DoubleResetDetector 
            - name: ArduinoJson 
            - name: ezTime 
            - name: UniversalTelegramBot
            - name: TFT_eSPI
            - name: PNGdec
          sketch-paths: |
            - F1-Notifications
          enable-warnings-report: true
          verbose: true
          cli-compile-flags: |
            - --export-binaries

      # Copy binaries to GitHubPages folder for publishing
      - name: Copy User_Setup.h
        run: |
          \cp -fR DisplayConfig/User_Setup.h ~/Arduino/libraries/TFT_eSPI/

      # Build Arduino Code
      - uses: arduino/compile-sketches@v1
        name: Compile CYD code
        with: 
          fqbn: "esp32:esp32:esp32"
          platforms: |
            - name: esp32:esp32
              source-url: https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json
          # No need to specify ESP-libraries as these are installed with the platform (on the line above). 
          # Downloading SpeedyStepper from source to get correct casing on import of Arduino.h (spelled arduino.h in the version from Arduino Library Manager)
          libraries: | 
            - name: ImageFetcher
              source-url: https://github.com/witnessmenow/file-fetcher-arduino.git
          sketch-paths: |
            - F1-Notifications
          enable-warnings-report: true
          verbose: true
          cli-compile-flags: |
            - --export-binaries
```

per1234, 2023-05-15:
Hi @witnessmenow. The compilation runs in the same environment as the rest of your workflow job, so you can prepare the environment as needed in advance. This means that if you need to make any modifications to dependencies, simply run those operations in workflow steps before the step that executes the `arduino/compile-sketches` action.

Although you could do it various ways, if you want to install the latest release version of the `TFT_eSPI` library in preparation for injecting your customizations into it, [an `arduino-cli lib install` command](https://arduino.github.io/arduino-cli/latest/commands/arduino-cli_lib_install/) is probably the most convenient approach. Arduino provides another action named [`arduino/setup-arduino-cli`](https://github.com/arduino/setup-arduino-cli) that simply installs **Arduino CLI** into the runner environment, after which you can use it in arbitrary shell commands. So you can do something like this:

```yaml
      - name: Install Arduino CLI
        uses: arduino/setup-arduino-cli@v1

      - name: Install TFT_eSPI library to prepare for modifications
        run: |
          arduino-cli lib install TFT_eSPI

      - name: Copy User_Setup.h
        run: |
          cp -fR DisplayConfig/User_Setup.h ~/Arduino/libraries/TFT_eSPI/

      - uses: arduino/compile-sketches@v1
        name: Compile CYD code
        with: 
          fqbn: "esp32:esp32:esp32"
          platforms: |
            - name: esp32:esp32
              source-url: https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json
          # No need to specify ESP-libraries as these are installed with the platform (on the line above). 
          libraries: | 
            - name: ImageFetcher
              source-url: https://github.com/witnessmenow/file-fetcher-arduino.git
            - name: WiFiManager
            - name: ESP_DoubleResetDetector 
            - name: ArduinoJson 
            - name: ezTime 
            - name: UniversalTelegramBot
            - name: PNGdec
          sketch-paths: |
            - F1-Notifications
          enable-warnings-report: true
          verbose: true
          cli-compile-flags: |
            - --export-binaries
```

Note that I did not include `TFT_eSPI` in the `libraries` input of the `arduino/compile-sketches` step because it had already been installed by the prior step.

If you have any questions or problems while using Arduino's GitHub Actions actions, or with setting up continuous integration of Arduino projects in general, you are welcome to post over on [Arduino Forum](https://forum.arduino.cc/). I'll be happy to provide assistance over there.

closed 2023-05-15 by per1234

witnessmenow, 2023-05-15:
Thank you @per1234 , much appreciated