arduino/compile-sketches #167: [AE-158] Library not included, when `source-path:` points to private repo

aliphys, 2023-08-31:
### Describe the problem

I want to check that the examples in library A compile correctly **prior to public release**. However, library A is dependant upon library B. Both libraries are private. 

The [compile-sketches-private.yml](https://github.com/per1234/.github/blob/18100c7928dd8a37fb519820d06d43d366560385/workflow-templates/compile-sketches-private.yml) workflow works for library B, but not library A.

### To reproduce

1. Create two private repos. Let's call them `LibA` and `LibB`. `LibB` has no external dependancies, outside what is inside the library manager. `LibA` however is dependant upon `LibB` (i.e. has `#include "LibB"` in the example sketches)
2. Copy the [compile-sketches-private.yml](https://github.com/per1234/.github/blob/18100c7928dd8a37fb519820d06d43d366560385/workflow-templates/compile-sketches-private.yml) workflow to the `.github/workflows` folder in each repo. Configure the `fqbn:` and add libraies from the library manager (e.g. `libraries: - name: Servo`)
3. Inside the `compile-sketches-private.yml` for `LibA`, refer to `LibB` as follows:
```
 libraries: |
            # Install the library from the local path.
            - source-path: ./
            - source-path: https://github.com/arduino-libraries/LibB
```
4. When running the workflow, I get a message stating the following:
```
Error: Library source path: https:/github.com/arduino-libraries/LibB doesn't exist
Error: Process completed with exit code 1.
```

### Expected behavior

Private libraries to which my account (or the organisation hosting it) have access to should be acceptable entries to `source-path:`

### 'arduino/compile-sketches' version

https://github.com/per1234/.github/commit/b7b3433b6144548ddba4b2e46b10b600ed336995

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/compile-sketches/issues?q=)
- [X] I verified the problem still occurs when using the latest version
- [X] My report contains all necessary details

aliphys, 2023-09-05:
In this issue, I requested that the Expected behaviour should be: Private libraries to which my account (or the organisation hosting it) have access to should be acceptable entries to _source-path:_

However, `source-path:` can only be used for location relative to the current repository location. The correct syntax (as pointed out by @sebromero ) is `source-url:`

This is also stated in the documentation. 

https://github.com/arduino/compile-sketches/blob/e841385c0a8a242a3f7254f1c9d9edefc27940ba/README.md?plain=1#L100-L116

Closing issue, as not relavent

closed 2023-09-05 by aliphys

per1234, 2023-09-05:
Thanks for your report @aliphys.

>The correct syntax (as pointed out by @sebromero ) is `source-url:`

This is true, but even with that error fixed in the workflow:

```yaml
libraries: |
  - source-url: https://github.com/arduino-libraries/LibB.git
```

the installation of the dependency would still fail if `arduino-libraries/LibB` is a private repository. Installation of dependencies from private repositories is supported by the action using the standard capabilities of Git, GitHub, and GitHub Actions, but the necessary configuration is fairly complex so I think it would be useful to provide some guidance about that. I have submitted a PR to add this information to the documentation: https://github.com/arduino/compile-sketches/pull/174

Since you have a valuable perspective on this subject as someone who was recently attempting to use the action in the targeted use case, I would appreciate it if you would review that PR.