arduino/compile-sketches #251: Cannot build a sketch for esp8266

UncleGrumpy, 2024-03-12:
### Describe the problem

The workflow sets up a python environment, with all of the necessary dependencies, but then tries to execute a completely different (and non existent) hard coded version of python to do the build:

```
Action setup
  creating virtual environment...
  installing poetry from spec 'poetry==1.4.0'...
  done! ✨ 🌟 ✨
    installed package poetry 1.4.0, installed using Python 3.11.2
    These apps are now globally available
      - poetry
  Creating virtualenv compilesketches-jE1jbGsw-py3.11 in /home/runner/.cache/pypoetry/virtualenvs
  Installing dependencies from lock file
  Package operations: 19 installs, 0 updates, 0 removals
    • Installing pycparser (2.21)
    • Installing cffi (1.15.1)
    • Installing certifi (2022.12.7)
    • Installing charset-normalizer (3.1.0)
    • Installing cryptography (40.0.1)
    • Installing idna (3.4)
    • Installing smmap (5.0.0)
    • Installing urllib3 (1.26.15)
    • Installing wrapt (1.15.0)
    • Installing deprecated (1.2.13)
    • Installing gitdb (4.0.10)
    • Installing pyjwt (2.6.0)
    • Installing pynacl (1.5.0)
    • Installing requests (2.28.2)
    • Installing gitpython (3.1.31)
    • Installing pygithub (1.58.1)
    • Installing pyserial (3.5)
    • Installing pyyaml (6.0)
    • Installing semver (2.13.0)
  Installing the current project: compilesketches (0.0.0)
Run # Run action
  # Run action
  poetry run \
    python compilesketches/compilesketches.py
  shell: /usr/bin/bash --noprofile --norc -e -o pipefail {0}
  env:
    UNIVERSAL_LIBRARIES: - name: Websockets
  - name: ESPAsyncTCP
  
    SKETCHES_REPORTS_PATH: sketches-reports
    pythonLocation: /opt/hostedtoolcache/Python/3.11.2/x64
    PKG_CONFIG_PATH: /opt/hostedtoolcache/Python/3.11.2/x64/lib/pkgconfig
    Python_ROOT_DIR: /opt/hostedtoolcache/Python/3.11.2/x64
    Python2_ROOT_DIR: /opt/hostedtoolcache/Python/3.11.2/x64
    Python3_ROOT_DIR: /opt/hostedtoolcache/Python/3.11.2/x64
    LD_LIBRARY_PATH: /opt/hostedtoolcache/Python/3.11.2/x64/lib
    INPUT_CLI-VERSION: latest
    INPUT_FQBN: esp8266:esp8266:esp8285
    INPUT_LIBRARIES: - name: Websockets
  - name: ESPAsyncTCP
  
    INPUT_PLATFORMS: - name: esp8266:esp8266
    source-url: https://github.com/esp8266/Arduino.git
    version: 3.1.2
  
    INPUT_SKETCH-PATHS: ./
    INPUT_CLI-COMPILE-FLAGS: 
    INPUT_VERBOSE: false
    INPUT_GITHUB-TOKEN: 
    INPUT_ENABLE-DELTAS-REPORT: true
    INPUT_ENABLE-WARNINGS-REPORT: true
    INPUT_SKETCHES-REPORT-PATH: sketches-reports
  
Compiling sketch: .
  Used platform   Version Path
  esp8266:esp8266 3.1.2   /home/runner/Arduino/hardware/esp8266/esp8266
  Error during build: fork/exec /home/runner/Arduino/hardware/esp8266/esp8266/tools/python3/python3: no such file or directory
Error: Compilation failed
Error: One or more compilations failed
Error: Process completed with exit code 1.
```

### To reproduce

Compile any sketch for esp8266.

### Expected behavior

I expect the workflow to see if my sketch builds, not fail before the build process has even begun.

### 'arduino/compile-sketches' version

arduino/compile-sketches@v1

### Additional context

This is my workflow yaml file, did I make a stupid mistake here?

```
name: Compile Sketch

# The workflow will run on Saturday's at 2:30am UTC & every push and pull request to the repository 
on:
  schedule:
    - cron: "30 2 * * 6"
  push:
  pull_request:

jobs:
  compile-sketch:
    runs-on: ubuntu-latest

    strategy:
      fail-fast: false

      matrix:
        board:
          - fqbn: esp8266:esp8266:nodemcu
          - fqbn: esp8266:esp8266:esp8285
        core:
          - version: 3.1.2
          - version: 3.0.2

    steps:
      - name: Checkout repository
        uses: actions/checkout@v4

      - name: Compile examples
        uses: arduino/compile-sketches@v1
        with:
          platforms: |
            - name: esp8266:esp8266
              source-url: https://github.com/esp8266/Arduino.git
              version: ${{ matrix.core.version }}
          fqbn: ${{ matrix.board.fqbn }}
          libraries: |
            - name: Websockets
            - name: ESPAsyncTCP
          sketch-paths: ./
```

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/compile-sketches/issues?q=)
- [X] I verified the problem still occurs when using the latest version
- [X] My report contains all necessary details

per1234, 2024-03-12:
Hi @UncleGrumpy.

## Explanation of Problem

This part of your workflow causes the "esp8266" platform to be installed from the [Git repository source](https://github.com/arduino/compile-sketches#repository):

>```yaml
>          platforms: |
>            - name: esp8266:esp8266
>              source-url: https://github.com/esp8266/Arduino.git
>              version: ${{ matrix.core.version }}
>```

When you use the standard way of installing the platform, all the tool dependencies are installed along with it via the Boards Manager infrastructure. But when you install it from a Git repository source, no tool dependencies are installed. One of the tool dependencies of the "esp8266" platform is Python. This is the cause of the error you encountered.

There is no possible way for the action to know the arbitrary procedure a platform would need for tool dependencies installation when installed from a repository source so you must configure your workflow to install all dependencies of the platform if you are going to source it from a repository.

## Recommended Solution

The best solution will be to simply [source the platform from Boards Manager](https://github.com/arduino/compile-sketches#boards-manager):

```yaml
          platforms: |
            - name: esp8266:esp8266
              source-url: https://arduino.esp8266.com/stable/package_esp8266com_index.json
              version: ${{ matrix.core.version }}
```

Sourcing from a Git repository is useful in cases where you want to test against the development version of a platform, but in this case you are only using the release versions of the platform so there is absolutely no benefit to sourcing it from the repository.

## Notes re: Using the Repository Source

When it does make sense to use a repository source for a platform, tool dependencies installation is typically accomplish that by installing the release version of the platform sourced from Boards Manager just to get the tool dependencies and then overwrite that installation with the repository sourced platform. You can see an example of that here:

https://github.com/arduino/ArduinoCore-avr/blob/63092126a406402022f943ac048fa195ed7e944b/.github/workflows/compile-platform-examples.yml#L186-L191

```yaml
          platforms: |
            # Use Boards Manager to install the latest release of the platform to get the toolchain.
            - name: arduino:avr
            # Overwrite the Boards Manager installation with the platform from the repository.
            - source-path: ./
              name: arduino:avr
```

However, I haven't ever tried that with the "esp8266" platform. The standard way to install the tool dependencies when using a manual installation of that platform is the procedure described here:

https://arduino-esp8266.readthedocs.io/en/latest/installing.html#using-git-version

Providing support for this sort of thing is out of scope for this repository. You can ask over on [**Arduino Forum**](https://forum.arduino.cc/) or the [ESP8266 Forum](https://www.esp8266.com/) if you need assistance with that.

closed 2024-03-12 by per1234

UncleGrumpy, 2024-03-12:
Thank you so much for your quick response! I had previously searched both forums, without finding a answer, before opening this issue.  I had originally used the package index url, but had another mistake in my workflow file, and in my confusion, switched to the manual package install (which I now realize would have failed to get the matrix versions correct anyway).

Just switching the url back to using the package index fixed all my problems. Thank you for this project! It will be a real life saver.