arduino/compile-sketches #263: Compile ESP8266 Platform examples => esp8266 installation fails

hasenradball, 2024-04-24:
### Describe the problem

I created a workflow file together with git copilot, and got this:

But it seems the is a mistake with the esp8266 installation.
The esp platform installation fails.

Can anyone help here?

```

name: Compile Examples

on: [push, pull_request]

jobs:
  build-arduino:
    runs-on: ubuntu-latest
    if: contains(github.event.head_commit.message, '[arduino]')
    steps:
      - uses: actions/checkout@v3
      - uses: arduino/compile-sketches@v1
        with:
          libraries: |
            - source-path: ./

  build-esp8266:
    runs-on: ubuntu-latest
    if: contains(github.event.head_commit.message, '[esp8266]')
    steps:
      - name: Checkout code
        uses: actions/checkout@v3

      - name: Install Arduino CLI
        uses: arduino/setup-arduino-cli@v1

      - name: Install ESP8266 platform
        run: |
          arduino-cli core update-index
          arduino-cli core install esp8266:esp8266

      - name: Compile code
        run: |
          arduino-cli compile --fqbn esp8266:esp8266:d1_mini ./examples/Wemos_D1_mini_AM2302_Sensor_Array_Example/Wemos_D1_mini_AM2302_Sensor_Array_Example.ino
          
```

```
builtin:serial-discovery@1.4.0 0 B / 1.63 MiB    0.00%
builtin:serial-discovery@1.4.0 downloaded
Installing builtin:serial-discovery@1.4.0...
Skipping tool configuration....
builtin:serial-discovery@1.4.0 installed

Downloading index: package_index.tar.bz2 0 B / 57.83 KiB    0.00%
Downloading index: package_index.tar.bz2 downloaded
Invalid argument passed: Found 0 platform for reference "esp8266:esp8266":

Error: Process completed with exit code 7.
```

### To reproduce

every time when an esp example is pushed

### Expected behavior

no error in installation

### 'arduino/compile-sketches' version

v1

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/compile-sketches/issues?q=)
- [X] I verified the problem still occurs when using the latest version
- [X] My report contains all necessary details

hasenradball, 2024-04-24:
I tried with the following;

```
name: Compile Examples

on: [push, pull_request]

jobs:
  build-arduino:
    runs-on: ubuntu-latest
    if: contains(github.event.head_commit.message, '[arduino]')
    steps:
      - uses: actions/checkout@v3
      - uses: arduino/compile-sketches@v1
        with:
          libraries: |
            - source-path: ./

  build-esp8266:
    runs-on: ubuntu-latest
    if: contains(github.event.head_commit.message, '[esp8266]')
    steps:
      - name: Checkout code
        uses: actions/checkout@v3

      - name: Install Arduino CLI
        uses: arduino/setup-arduino-cli@v1
        
      - name: Config additional Boards for ESP8266 platform
        run: |
          arduino-cli config add board_manager.additional_urls http://arduino.esp8266.com/stable/package_esp8266com_index.json

      - name: List ESP8266 Boards
        run: |
          arduino-cli core search esp8266
          
      - name: Install ESP8266 platform
        run: |
          arduino-cli core install esp8266:esp8266

      - name: Compile code
        run: |
          arduino-cli compile --fqbn esp8266:esp8266:d1_mini ./examples/Wemos_D1_mini_AM2302_Sensor_Array_Example/Wemos_D1_mini_AM2302_Sensor_Array_Example.ino

```

But failed also:

How does it work to install additional Boards in arduino workflow?

per1234, 2024-04-24:
Hi @hasenradball. This issue tracker is only to be used for reporting bugs with the **arduino/compile-sketches** GitHub Actions action, or submitting feature requests for the action. For assistance with setting up GitHub Actions workflows for your Arduino projects, please use Arduino Forum:

https://forum.arduino.cc/

We'll be happy to help you over there.

In addition, your problem is related to using **Arduino CLI** directly in the workflow, rather than the the **arduino/compile-sketches** GitHub Actions action so it is also off topic for this repository.

closed 2024-04-24 by per1234