arduino/compile-sketches #284: compilesketches.py error before compile that related to recent update

M-ichae-l, 2024-06-13:
### Describe the problem

I have use this action to do CI check for all examples. However, it gives me errors fund in py files. I have tried last week passed action to re-run the error also comes out.  I guess it related with some code recent update used by this GitHub actions. I have tried with main and v1.1.0, neither can work. Found the arduino-cli has an update today. Maybe it is the update cause the issue. 

> File "/home/runner/work/_actions/arduino/compile-sketches/main/compilesketches/compilesketches.py", line 563, in get_platform_installation_path
>     if installed_platform[self.cli_json_key("core list", "ID")] == platform[self.dependency_name_key]:
>        ~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
> TypeError: string indices must be integers, not 'str'

### To reproduce

basic/default use of the action

### Expected behavior

Can run compilesketches with no process error.  

### 'arduino/compile-sketches' version

main / v1.1.0

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/compile-sketches/issues?q=)
- [X] I verified the problem still occurs when using the latest version
- [X] My report contains all necessary details

closed 2024-06-13 by per1234

per1234, 2024-06-13:
Thanks for your report @M-ichae-l. As you guessed, this was caused by breaking changes that were made in the 1.0.0 release of Arduino CLI. [By default](https://github.com/arduino/compile-sketches#cli-version), the **arduino/compile-sketches** uses the latest release of Arduino CLI so the workflows automatically started using Arduino CLI 1.0.0 as soon as that release came out.

The bug has been fixed and shipped via a new 1.1.1 release. I submitted a pull request to update the version of the action in your workflow: https://github.com/ambiot/ambpro2_arduino/pull/246

M-ichae-l, 2024-06-14:
Thank you! Many thanks for the fix.