arduino/compile-sketches #310: Attempting to use arduino:avr git repository as source-url results in compilesketches.py crashing

noisymime, 2024-08-12:
### Describe the problem

Trying to use the the latest AVR core from github results in `compilesketches.py` crashing with the following error:

```
 Traceback (most recent call last):
  File "/home/runner/work/_actions/arduino/compile-sketches/v1.1.1/compilesketches/compilesketches.py", line 1709, in <module>
Installing Arduino CLI version latest
Installing platform from repository: https://github.com/arduino/ArduinoCore-avr.git
    main()  # pragma: no cover
    ^^^^^^
  File "/home/runner/work/_actions/arduino/compile-sketches/v1.1.1/compilesketches/compilesketches.py", line 63, in main
    compile_sketches.compile_sketches()
  File "/home/runner/work/_actions/arduino/compile-sketches/v1.1.1/compilesketches/compilesketches.py", line 219, in compile_sketches
    self.install_platforms()
  File "/home/runner/work/_actions/arduino/compile-sketches/v1.1.1/compilesketches/compilesketches.py", line 337, in install_platforms
    self.install_platforms_from_repository(platform_list=platform_list.repository)
  File "/home/runner/work/_actions/arduino/compile-sketches/v1.1.1/compilesketches/compilesketches.py", line 629, in install_platforms_from_repository
    destination_path = self.get_platform_installation_path(platform=platform)
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/runner/work/_actions/arduino/compile-sketches/v1.1.1/compilesketches/compilesketches.py", line 562, in get_platform_installation_path
    for installed_platform in installed_platform_list:
TypeError: 'NoneType' object is not iterable
Error: Process completed with exit code 1.
```

### To reproduce

Use the following configuration taken nearly verbatim from the FAQ page (https://github.com/arduino/compile-sketches/blob/main/docs/FAQ.md):

```
- uses: arduino/compile-sketches@v1.1.1
  with:
    platforms: |
      - source-url: https://github.com/arduino/ArduinoCore-avr.git
        name: arduino:avr
```

### Expected behavior

If the `source-url` field is removed from the above example then everything works as expected. This should work identically with the github repository as the source as it does with the build in AVR core. 

### 'arduino/compile-sketches' version

1.1.1

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/compile-sketches/issues?q=)
- [X] I verified the problem still occurs when using the latest version
- [X] My report contains all necessary details

closed 2024-08-13 by per1234

per1234, 2024-08-13:
Thanks for your report @noisymime. The bug has now been fixed in https://github.com/arduino/compile-sketches/pull/311 and that fix shipped in a new v1.1.2 release of the action.

Since you are currently pinning the action version, in order to get the fix in your workflow, you must change this line:

```yaml
- uses: arduino/compile-sketches@v1.1.1
```

To this:

```yaml
- uses: arduino/compile-sketches@v1.1.2
```

noisymime, 2024-08-13:
> The bug has now been fixed in #311 and that fix shipped in a new v1.1.2 release of the action.

You legend! Thanks for looking at this so quickly @per1234 !