arduino/compile-sketches #314: Error creating report file

thomascenni, 2024-08-30:
### Describe the problem

When using a complex fqbn like this one:

fqbn:esp32:esp32:esp32s3:UploadSpeed=921600,USBMode=hwcdc,CDCOnBoot=default,MSCOnBoot=default,DFUOnBoot=default,UploadMode=default,CPUFreq=240,FlashMode=qio,FlashSize=16M,PartitionScheme=app3M_fat9M_16MB,DebugLevel=none,PSRAM=opi,LoopCore=1,EventsCore=0,EraseFlash=none,JTAGAdapter=default,ZigbeeMode=default

the report file name is too long and cannot be saved.
Should add a parameter with the name for the report.

[image]

### To reproduce

Use fqbn provided.

### Expected behavior

File saved.

### 'arduino/compile-sketches' version

1.1.2

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/compile-sketches/issues?q=)
- [X] I verified the problem still occurs when using the latest version
- [X] My report contains all necessary details

per1234, 2024-08-31:
Hi @thomascenni. Thanks for your report. We will try to look into adjusting the code to handle this when we have time.

Until then, you should shorten the FQBN as a workaround. When you don't specify a custom board option in the FQBN, the default (which is the first option on the list) will be used. This means you only need to specify custom board options when the default value is not suitable. The only options in your FQBN that aren't defaults are the following:

- `FlashSize=16M`
- `PartitionScheme=app3M_fat9M_16MB`
- `PSRAM=opi`
- `EventsCore=0`

So you can use this FQBN:

```text
esp32:esp32:esp32s3:FlashSize=16M,PartitionScheme=app3M_fat9M_16MB,PSRAM=opi,EventsCore=0
```

thomascenni, 2024-08-31:
Hello @per1234 
Thanks for the suggestion, that's exactly what I did as workaround.
Is it possible to use the settings (fqbn, libraries, ...) already defined in the https://arduino.github.io/arduino-cli/1.0/sketch-project-file/ ?

per1234, 2024-08-31:
Unfortunately there isn't any support for using the action with the sketch project file. We are tracking the request for adding such support here: https://github.com/arduino/compile-sketches/issues/248