arduino/compile-sketches #333: requests 2.32.0 is yanked, cannot build.

SJang1, 2024-11-26:
### Describe the problem

I see that there was version changes on #274 which did made requests 2.32.0, however it seems to be like this version have a CVE issue that made blocking the project to build.

Gets error on Actions setup process

### To reproduce

Set yaml to use `uses: arduino/compile-sketches@v1.1.1` or higher version.
current latest (1.1.2) still uses requests 2.32.0 and it makes build unavailable

### Expected behavior

Build builds correctly

### 'arduino/compile-sketches' version

1.1.1 or 1.1.2 or latest

### Additional context

```
  creating virtual environment...
  installing poetry from spec 'poetry==1.4.0'...
  done! ✨ 🌟 ✨
    installed package poetry 1.4.0, installed using Python 3.11.2
    These apps are now globally available
      - poetry
  Creating virtualenv compilesketches--acYsnh9-py3.11 in /home/runner/.cache/pypoetry/virtualenvs
  Installing dependencies from lock file
  Package operations: 20 installs, 0 updates, 0 removals
    • Installing pycparser (2.21)
    • Installing cffi (1.15.1)
    • Installing certifi (2023.7.22)
    • Installing charset-normalizer (3.1.0)
    • Installing cryptography (42.0.4)
    • Installing idna (3.7)
    • Installing smmap (5.0.0)
    • Installing urllib3 (1.26.18)
    • Installing wrapt (1.15.0)
    • Installing deprecated (1.2.13)
    • Installing gitdb (4.0.10)
    • Installing pyjwt (2.6.0)
    • Installing pynacl (1.5.0)
    • Installing requests (2.32.0)
    • Installing typing-extensions (4.8.0)
    • Installing gitpython (3.1.43)
    • Installing pygithub (2.3.0)
    • Installing pyserial (3.5)
    • Installing semver (3.0.2)
    • Installing pyyaml (6.0.1)
  Warning: The file chosen for install of requests 2.32.0 (requests-2.32.0-py3-none-any.whl) is yanked. Reason for being yanked: Yanked due to conflicts with CVE-2024-35195 mitigation
  Installing the current project: compilesketches (0.0.0)
```

### Issue checklist

- [x] I searched for previous reports in [the issue tracker](https://github.com/arduino/compile-sketches/issues?q=)
- [x] I verified the problem still occurs when using the latest version
- [x] My report contains all necessary details

per1234, 2024-11-26:
Hi @SJang1. This is only a warning, not an error. Your "it makes build unavailable" claim is false. Arduino uses the action in hundreds of libraries and it is fully functional 

Are you actually experiencing a problem while using the **arduino/compile-sketches** action? If so, please provide a detailed description of the problem, including instructions I can follow to reproduce it. If you are using the action in a workflow in a public repository, it will be helpful if you provide a link to a workflow run in the repository that demonstrates the problem.

SJang1, 2024-11-26:
It actually blocks from building, when I clone the repo and change the version of it up, change uses to my repo@main and run, it successes.

SJang1, 2024-11-26:
Oh, founf that it can keep going done by just warning, I just found out another problem, and seems like there was my mistake.

Thanks!

closed 2024-11-26 by SJang1