arduino/compile-sketches #368: cli-compile-flags not working

macsnoeren, 2025-03-25:
### Describe the problem

I try to enable that the build files are exported to the sketch directory. Unfortunatuly, I could not debug the problem at this moment.

### To reproduce

The action file:
```
name: build
on: [push, pull_request]
jobs:
  compile-sketches:
    runs-on: ubuntu-latest
    steps:
      - uses: actions/checkout@v4  # Updated to latest version
      
      - name: Compile Arduino Sketches
        uses: arduino/compile-sketches@v1
        with:
          verbose: true
          fqbn: esp8266:esp8266:d1_mini_pro
          platforms: |
            - source-url: https://arduino.esp8266.com/stable/package_esp8266com_index.json
              name: esp8266:esp8266
          sketch-paths: |
            - Firmware/sm-esp8266
          libraries: |
            - name: WiFiManager
            - name: ArduinoJson
            - name: PubSubClient
          cli-compile-flags: "-e"

      - name: List Files
        run: |
          ls -l Firmware/sm-esp8266
```

This gives the result at the compilation step that the arduino-cli is not correctly called:
```
Compiling sketch: Firmware/sm-esp8266
  Error: accepts at most 1 arg(s), received 3
  Usage:
    arduino-cli compile [flags]
  
  Examples:
    /home/runner/bin/arduino-cli compile -b arduino:avr:uno /home/user/Arduino/MySketch
    /home/runner/bin/arduino-cli compile -b arduino:avr:uno --build-property "build.extra_flags=\"-DMY_DEFINE=\"hello world\"\"" /home/user/Arduino/MySketch
    /home/runner/bin/arduino-cli compile -b arduino:avr:uno --build-property "build.extra_flags=-DPIN=2 \"-DMY_DEFINE=\"hello world\"\"" /home/user/Arduino/MySketch
    /home/runner/bin/arduino-cli compile -b arduino:avr:uno --build-property build.extra_flags=-DPIN=2 --build-property "compiler.cpp.extra_flags=\"-DSSID=\"hello world\"\"" /home/user/Arduino/MySketch
  
  
  Flags:
        --board-options strings                 List of board options separated by commas. Or can be used multiple times for multiple options.
        --build-path string                     Path where to save compiled files. If omitted, a directory will be created in the default temporary path of your OS.
        --build-property stringArray            Override a build property with a custom value. Can be used multiple times for multiple properties.
        --clean                                 Optional, cleanup the build folder and do not use any cached build.
        --discovery-timeout duration            Max time to wait for port discovery, e.g.: 30s, 1m (default 1s)
        --dump-profile                          Create and print a profile configuration from the build.
        --encrypt-key string                    The name of the custom encryption key to use to encrypt a binary during the compile process. Used only by the platforms that support it.
    -e, --export-binaries                       If set built binaries will be exported to the sketch folder.
    -b, --fqbn string                           Fully Qualified Board Name, e.g.: arduino:avr:uno
    -h, --help                                  help for compile
    -j, --jobs int32                            Max number of parallel compiles. If set to 0 the number of available CPUs cores will be used.
        --keys-keychain string                  The path of the dir to search for the custom keys to sign and encrypt a binary. Used only by the platforms that support it.
        --libraries strings                     Path to a collection of libraries. Can be used multiple times or entries can be comma separated.
        --library strings                       Path to a single library’s root folder. Can be used multiple times or entries can be comma separated.
        --only-compilation-database             Just produce the compilation database, without actually compiling. All build commands are skipped except pre* hooks.
        --optimize-for-debug                    Optional, optimize compile output for debugging, rather than for release.
        --output-dir string                     Save build artifacts in this directory.
    -p, --port string                           Upload port address, e.g.: COM3 or /dev/ttyACM2
        --preprocess                            Print preprocessed code to stdout instead of compiling.
    -m, --profile string                        Sketch profile to use
    -P, --programmer string                     Programmer to use, e.g: atmel_ice
    -l, --protocol string                       Upload port protocol, e.g: serial
    -q, --quiet                                 Optional, suppresses almost every output.
        --show-properties string[="expanded"]   Show build properties. The properties are expanded, use "--show-properties=unexpanded" if you want them exactly as they are defined. (default "disabled")
        --sign-key string                       The name of the custom signing key to use to sign a binary during the compile process. Used only by the platforms that support it.
    -u, --upload                                Upload the binary after the compilation.
    -v, --verbose                               Optional, turns on verbose mode.
    -t, --verify                                Verify uploaded binary after the upload.
        --warnings string                       Optional, can be: none, default, more, all. Used to tell gcc which warning level to use (-W flag). (default "none")
  
  Global Flags:
        --additional-urls strings   Comma-separated list of additional URLs for the Boards Manager.
        --config-dir string         Sets the default data directory (Arduino CLI will look for configuration file in this directory).
        --config-file string        The custom config file (if not specified the default will be used).
        --json                      Print the output in JSON format.
        --log                       Print the logs on the standard output.
        --log-file string           Path to the file where logs will be written.
        --log-format string         The output format for the logs, can be: text, json (default "text")
        --log-level string          Messages with this level and above will be logged. Valid levels are: trace, debug, info, warn, error, fatal, panic (default "info")
        --no-color                  Disable colored output.
  
  accepts at most 1 arg(s), received 3
```

### Expected behavior

That the compilation is done correctly, which is when I remove the 'cli-compile-flags', and that the binary files are found in the sketch directory.

### 'arduino/compile-sketches' version

latest

### Issue checklist

- [x] I searched for previous reports in [the issue tracker](https://github.com/arduino/compile-sketches/issues?q=)
- [x] I verified the problem still occurs when using the latest version
- [x] My report contains all necessary details

macsnoeren, 2025-03-27:
The code responsible for the compile code, can be found at row 912. From row 922 to 925 the compilation flags are created. 
```
   def compile_sketch(self, sketch_path, clean_build_cache):

                ............

        compilation_command = ["compile", "--warnings", "all", "--fqbn", self.fqbn]
        if self.cli_compile_flags is not None:
            compilation_command.extend(self.cli_compile_flags)
        compilation_command.append(sketch_path)
```

per1234, 2025-03-28:
Hi @macsnoeren. Because the input supports specifying multiple arguments, the value of the `cli-compile-flags` input must be formatted as a YAML sequence (array).

Let's start by looking at the input value by itself. Your current value is:

```yaml
-e
```

The data type of this YAML document is a string.

This is how to convert it to a sequence type:

```yaml
- -e
```

Now that we understand how to correctly format the input value, we need to apply this change to the workflow. If GitHub Actions supported sequence type action input values, we could do something like this:

```yaml
cli-compile-flags:
  - -e
```

However, GitHub Actions only supports the string type for action input values. So we use a string that is in turn a YAML document (the action takes care of parsing it as such). Making the most minimal change to your existing code, this would look like:

```yaml
cli-compile-flags: "- -e"
```

However, I think a more readable form is this:

```yaml
cli-compile-flags: |
  - -e
```

This uses the "[literal block scalar style](https://www.yaml.info/learn/quote.html#literal)" (note the `|` indicator). This style is especially superior in cases where you want to specify multiple arguments.

>Another improvement could be that the actual call is printed to the output

I think it is an excellent suggestion. Please open a dedicated feature request so that the task can be tracked effectively.

closed 2025-03-28 by per1234